jflight-public/jesc-configurator #2: RPM feedback not working

chuckwain, 2019-10-11:
I just loaded BetaFlight 4.1 and the JESC BLHeli firmware on my Larva X.  The Larva X runs the CRAZYBEEF4DX flight controller with integrated ESC's.  I have followed the instructions to enable bidirectional DSHOT but I am not getting any rpm data coming back from the ESC's in the BetaFlight configurator.  I have this setup working on two other quads that have BLHeli 32 ESC's so I have experience.  I found some reference to a timer issue on this flight controller but have had no luck getting to work.  Here are my BetaFlight settings:

# get dshot
dshot_idle_value = 550
Allowed range: 0 - 2000

dshot_burst = OFF
Allowed values: OFF, ON

dshot_bidir = ON
Allowed values: OFF, ON
Default value: OFF

dshot_bitbang = AUTO
Allowed values: OFF, ON, AUTO

dshot_bitbang_timer = AUTO
Allowed values: AUTO, TIM1, TIM8

beeper_dshot_beacon_tone = 3
Allowed range: 1 - 5
Default value: 1

Is there a solution to this?

chuckwain, 2019-10-11:
I found the issue.  There are steps to loading the JFLIGHT firmware.  First the base firmware must be upgraded in the JESC Configurator, then you have to go back and load the telemetry firmware.  It is a two step process.  Now it works!  This was NOT well documented, I found it on a unrelated website online.

joelucid, 2019-10-18:
@chuckwain the [Install Instructions](https://jflight.net/index.php?route=information/information&language=en-gb&information_id=7) lay it out pretty clearly.

closed 2019-10-18 by joelucid