jflight-public/jesc-configurator #3: ignore inappropriate MCU cant pass the flash all step

DarkF4lcon, 2019-10-17:
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/3741064/Log.txt)

Cant pass the step flash all

joelucid, 2019-10-18:
It looks like this is a L style esc which isn't supported in JESC. Could you read setup in blhelis-configurator and send a screenshot?

DarkF4lcon, 2019-10-18:
Here it is

What can i do?

[image]
[image]

joelucid, 2019-10-21:
Could you try to reflash these to 16.7 in blheli configurator let me know if that works. If it does then retry to flash jesc.

DarkF4lcon, 2019-10-21:
Reflashed the escs but it doesnt work neither
Is it possible to reset my licences for me to try on my other drone?
I have a 4in1 esc from Mamba on the other one

joelucid, 2019-11-14:
Ok - with the escs flashed back to blhelis A-H-20 please try to flash jesc again and then post the jesc-configurator log.

DarkF4lcon, 2019-11-14:
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/3849111/Log.txt)
Here it is, Reflashed the ESCs again, Still nothing to select after clicking on Flash all in JESC

joelucid, 2019-11-15:
Ok - the signature returned by the boot loader of the esc is 0xe8b1:
```
Uncaught Error: Unknown MCU signature: e8b1 chrome-extension://maigancljkhnkhpejlfmgjjmjafafdhp/js/build/firmware_selector.js 26 13 Error: Unknown MCU signature: e8b1
```

This signature is used by L style escs only which jesc does not support. On the other hand the layout tag A-H-20 suggests it is a H style esc. One of the two is clearly wrong.

Could you please identify the type of cpu on the esc? Make a pic of the esc at a high resolution  and angle that the chip markings can be seen and post it here. What brand/model of esc is this?

If it is in fact an H style esc with the efm8bb21 you would likely need to patch the bootloader to fix it. This could be done by flashing the escs with jesc or blhelis using the c2 interface. The normal bootloader interface jesc-configurator uses protects against modifying the bootloader for obvious reasons.

DarkF4lcon, 2019-11-15:
[image]
[image]
Thanks for the reply

Heres the photos of the esc type that i use in this drone.
Says BB21 and its a Racestar MS25A

What can i do to revert this situation?

Thanks

joelucid, 2019-11-15:
Yeah it's definitely a bb21 with incorrectly flashed bootloader. I've made a change to the server side configuration of jesc-configurator to allow this one to be flashed as well. Could you try again. please?

DarkF4lcon, 2019-11-15:
Hi there

It worked, thanks for fixing this situation

Thanks

joelucid, 2019-11-15:
Great @darkf4icon! Glad we got it sorted out!

closed 2019-11-15 by joelucid