jflight-public/jesc-configurator #4: ESC not responding after successful flashing

StillVoidingWarranties, 2019-10-20:
I followed the instructions step by step and flashing JESC 2.1 and Telemetry 2.1 firmware seemed to have worked. However, after I powercycled (nice new startup tones) 
and connected with betaflight configurator motor 4 showed 100% errors. I could not get to read the configuration of ESC 4 with JESC Configurator, BLHeli Configurator or BLHeli Suite since.
ESCs in question are Racerstar 40A BLHeli_S, type identifies as G-H-30.

[JESC Amax Log.txt](https://github.com/jflight-public/jesc-configurator/files/3748416/JESC.Amax.Log.txt)

joelucid, 2019-10-21:
Your log shows all escs being read without problem in JESC Configurator. Did you record it before motor 4 stopped working?

Please try this: switch bidirectional dshot off and check if all motors spin then. If they do then reenable bidirectional dshot and paste output of the cli commands "diff" and "resource show all".

StillVoidingWarranties, 2019-10-21:
The logfile is from the session I flashed JESC. Immediately after that I connected to Betaflight and saw that ESC 4 stopped working. Switching bidirectional DSHOT off did not change anything. I will check the cabling and resource assignment. I attached the output of diff and resource at the settings I tried (with bidirectional DSHOT off).
[BTFL_cli_Amax_20191021_resource.txt](https://github.com/jflight-public/jesc-configurator/files/3750352/BTFL_cli_Amax_20191021_resource.txt)
[BTFL_cli_Amax_20191021_diff.txt](https://github.com/jflight-public/jesc-configurator/files/3750353/BTFL_cli_Amax_20191021_diff.txt)

StillVoidingWarranties, 2019-10-21:
After removing the heat shrink from the ESC I found it to be in bad shape with some capacitors being ripped off. The quad did fly perfectly and motors spun nicely on the bench before flashing but there is a good chance JESC was not, or at least not the primary reason for the ESC becoming unresponsive.

joelucid, 2019-10-21:
Ok thanks. The log shows perfect success when flashing. I would check cabling first - a couple of people have had issues in that area. And then of course escs sometimes do die.

If the fw on the esc is corrupted - which is unlikely since the flash succeeded and was verified - you can reflash using the c2 interface. There's a how to here: https://www.youtube.com/watch?v=b-7VFt3IhhE

StillVoidingWarranties, 2019-10-21:
I swapped the ESC. In JESC I can see the 3 licensed ESC and the one unlicensed. What happens when I click "Activate All"? Do I spend 1 license or 4?

joelucid, 2019-10-21:
It should cost 1 license.

StillVoidingWarranties, 2019-10-21:
Thank you for the quick answer. Everything went fine, just 1 license was deducted and I now do get RPM info back on all ESCs (yey!). Sorry for bothering you, my ESC dying right after flashing was a strange coincidence that lead me to wrong assumptions. :-)

closed 2019-11-14 by joelucid