jflight-public/jesc-configurator #5: Unable to update A-H-20 ESCs

denvelt, 2019-10-23:


denvelt, 2019-10-23:
[JESCDebugLog.txt](https://github.com/jflight-public/jesc-configurator/files/3762180/JESCDebugLog.txt)

I am trying to update my old F3 build with BlheliS A-H-20 ESCs to run RPM filters with your BFPE firmware but I am unable to update my ESCs to JESC.. When I click the flash all nothing happens and the configurator disconnect from my quad...

DarkF4lcon, 2019-10-26:
Same here but im with Kakute F7

denvelt, 2019-10-26:
> Same here but im with Kakute F7
Was your license  reset? or are we screwed for following instructions???

DarkF4lcon, 2019-10-26:
Im waiting for instructions

DarkF4lcon, 2019-11-07:
> > Same here but im with Kakute F7
> > Was your license  reset? or are we screwed for following instructions???

Did you get it working?

denvelt, 2019-11-13:
> Im waiting for instructions

Did you ever get instructions?

Havn't had any reply for our issue...

joelucid, 2019-11-14:
@denvelt according to the log you shared the fw has an incorrect MCU signature. It claims to be an L style esc. This could be a mistake by the vendor or they could have done it on purpose. Could you verify that the MCU is actually an efm8bb21? You can do this by using it with dshot600 which is only supported by efm8bb21 cpus.

@DarkF4lcon IIRC you had the same issue? Could you also verify the MCU and post a log?

joelucid, 2019-11-15:
@denvelt what's the make and model of your esc?

joelucid, 2019-11-15:
Aha! Found some info on this: https://community.micro-motor-warehouse.com/t/racerstar-ms-series-esc-15a-repost/4007

Note this comment by @DieHertz: 
> BLHeli Suite is more strict in checking what firmware you have on your ESCs vs. what you’re trying to flash. Some ESCs have proper firmware but inappropriate bootloader which identifies then as BB1 when they are BB2. Hence users struggle to flash their ESCs using suite, still it’s possible to flash. Configurator checks firmware ESC indy instead of bootloader hardcoded info. Use whatever you like.

This is exactly what I'm seeing: bootloader says it's a efm8bb1, layout tag says efm8bb21. Now we only need to find out which is true. And then determine how to fix the problem.

denvelt, 2019-11-16:
@joelucid I was just able to flash my ESCs, My ESCs are the Racerstar MS 35A.  Thank you for helping us with this issue. Really appreciate what you do for the FPV community..

closed 2019-11-16 by denvelt