jflight-public/jesc-configurator #7: no read setup button on ubuntu

fractalfederation, 2019-11-05:
downloaded 1.2.5 and ran, it loads fine, able to connect but there is no read setup write setup buttons at bottom. only error i see in terminal is:

[29237:29237:1105/114454.969897:ERROR:component_loader.cc(165)] Failed to parse extension manifest.

not sure if its related.

TheIsotopes, 2019-11-05:
> downloaded 1.2.5 and ran, it loads fine, able to connect but there is no read setup write setup buttons at bottom.

same with both windows versions :( 

@joelucid can you check whats wrong.

joelucid, 2019-11-05:
I will look into it. Thanks for reporting!

joelucid, 2019-11-05:
@fractalfederation, @TheIsotopes which operating systems did you use?

Burnich, 2019-11-05:
Same problem on Windows 10 64 bit version

Burnich, 2019-11-05:
Same problem on Windows 10 64 bit version

> downloaded 1.2.5 and ran, it loads fine, able to connect but there is no read setup write setup buttons at bottom. only error i see in terminal is:
> 
> [29237:29237:1105/114454.969897:ERROR:component_loader.cc(165)] Failed to parse extension manifest.
> 
> not sure if its related.

Same problem on Windows 10 64bits version

TheIsotopes, 2019-11-05:
@joelucid using Windows 10 (1909 - 64bit)

Burnich, 2019-11-05:
I tried all the versions of the configurator (from 1.2.0 to 1.2.5) on Windows 10 64bit. Always the same result : I can connect but no read button...

joelucid, 2019-11-05:
Ok - downloaded a Windows 10 64bit VM and I can reproduce it.

fractalfederation, 2019-11-05:
ubuntu 18.04

joelucid, 2019-11-05:
These issues should now be solved.

fractalfederation, 2019-11-05:
built on ubuntu 18.04 and working! thanks

TheIsotopes, 2019-11-06:
> These issues should now be solved.

@joelucid can you release a new build to test this on windows 10

Burnich, 2019-11-06:
> > These issues should now be solved.
> 
> @joelucid can you release a new build to test this on windows 10

No need ! In my case I just re-download the 1.2.5 version after @joelucid last post and everything was fix !

joelucid, 2019-11-06:
@TheIsotopes you don't even need to update. I did respond 1.2.5 but the file where the error was is first attempted to download from jflight.net so I could just patch it there.

closed 2019-11-14 by joelucid