jflight-public/jesc-configurator #8: Same issue using Windows 7

blheliFan, 2019-11-08:
Im using Windows 7 64bit OS. My esc is hakrc 40A. Done flashing esc firmware jesc G40 48 2.2. Hex and g40 24 2.2 hex. But the configurator jesc still not read no menu.. 
[image]

blheliFan, 2019-11-08:
Im using Windows 7 64bit OS. My esc is hakrc 40A. Done flashing esc firmware jesc G40 48 2.2. Hex and g40 24 2.2 hex. But the configurator jesc still not read no menu.. 
[image]
[image]

blheliFan, 2019-11-08:
Im using Windows 7 64bit OS. My esc is hakrc 40A. Done flashing esc firmware jesc G40 48 2.2. Hex and g40 24 2.2 hex. But the configurator jesc still not read no menu.. 
[image]

joelucid, 2019-11-14:
This is now fixed, right @blheliFan? There was a short period where a server was misconfigured causing this.

closed 2019-11-15 by joelucid