jflight-public/jesc-configurator #10: Wont recognize ECS 4

Skyfall2, 2019-11-14:
Hi JESC license was recognized on 3 of my 4 ESC. All 4 ESC are recognized on BLHeli_S prior to loading on JESC. What is the next step from here? I have the debug files saved but not sure were to place it. Thanks - Craig

joelucid, 2019-11-15:
@Skyfall2 please upload the log here. You can just drag & drop it in.

Skyfall2, 2019-11-15:
[HAWK5.txt](https://github.com/jflight-public/jesc-configurator/files/3852120/HAWK5.txt)

Skyfall2, 2019-11-17:
[HAWK5.txt](https://github.com/jflight-public/jesc-configurator/files/3856299/HAWK5.txt)

joelucid, 2019-11-18:
Hey @Skyfall2. I just looked through the log in detail. Unfortunately your escs were locked by the manufacturer. Therefore jesc couldn't be installed. jesc-configurator 1.2.6 contains code to avoid attempting an install on such escs.

You will need to reflash the escs using the c2 interface to unlock them. JB goes over how this works here: https://www.youtube.com/watch?v=b-7VFt3IhhE

camoffitt, 2019-12-09:
Last 4in1 i did I lost esc #1, is brand new. What can I do?

Skyfall2, 2019-12-09:
Unfortunately your escs were locked by the manufacturer. Therefore jesc
couldn't be installed. jesc-configurator 1.2.6 contains code to avoid
attempting an install on such escs.

You will need to reflash the escs using the c2 interface to unlock them. JB
goes over how this works here: https://www.youtube.com/watch?v=b-7VFt3IhhE

camoffitt, 2019-12-09:
Thx. I’ll try. All my other escs were fine doing the rpm upgrade only this one failed.

kayHirano, 2019-12-10:
Hello,  One of ESC doesn't recognize on my AIO ESC board.
I attach the log on here.  I'm so glad if you can help me!

kayHirano, 2019-12-10:
[kay_log.txt](https://github.com/jflight-public/jesc-configurator/files/3943051/kay_log.txt)

joelucid, 2019-12-14:
@Skyfall2 is your issue now solved?

@camoffitt this can have multiple causes. Of you used 1.2.6 a locked esc would not be licensable at all. I need a jesc-configurator log to diagnose. Please open a new issue.

camoffitt, 2019-12-14:
No

Skyfall2, 2019-12-14:
Not yet. I have not had a chance to work on it yet. It is on my long list
of things to do. I'll follow what JB video says. Thanks for your help.

Craig