jflight-public/jesc-configurator #12: Wont recognize ECS

kayHirano, 2019-12-10:
Hello, One of ESC doesn't recognize on my AIO ESC board.
I attach the log on here. I'm so glad if you can help me!

my setup is  as follow

BetaFlight Performance Edition F3 4.1.1 EA3
target
betaflight_4.1.1_SPRACINGF3_FRSKY.hex
ECS original firmware
C-H-50

thanks
k
[kay_log.txt](https://github.com/jflight-public/jesc-configurator/files/3944632/kay_log.txt)

joelucid, 2019-12-11:
The log shows that even when you first read the esc configuration (prior to flashing jesc) esc 1 did not respond. Can you spin the motors in the motors tab?

kayHirano, 2019-12-11:
No I cant spin the motor in the motor tab.

joelucid, 2019-12-11:
Please try flashing the previous BF version you had on there and check if that works. Maybe there is an incompatibility of your AIO with 4.1.1 performance edition. If not your esc seems to have died before you had a chance to flash it with jesc.

kayHirano, 2019-12-11:
I tried to flash the previous Version as follow

BetaFlight Performance Edition F3 4.1 EA2
BetaFlight Performance Edition F3 4.1 EA1

However, the ESC 1did not recognized and cant spin the motor1 The ESC seems to die.
[debugLog.txt](https://github.com/jflight-public/jesc-configurator/files/3948761/debugLog.txt)

joelucid, 2019-12-14:
And @kayHirano are thesethe BF versions that previously did work with the FC? There are two possibilities here: either the new version of BF doesn't work with your setup or the esc died before you first read settings with jesc-configurator.

kayHirano, 2019-12-17:
Hi Joe, unfortunately, my ESC#1 on my 4in1 board seems to be dead. other 3ESCs are functional without any issue ironically.  I attempt to a physical approach as well as JB , however, my ESC #1 couldn't save.  
PS I use JESC firmware for another 3kwad s without any issue. I'm proud of your works!
thank you so much.

joelucid, 2019-12-17:
@kayHirano - thanks for the feedback. Yeah sometimes escs just die. :(

closed 2019-12-17 by joelucid