jflight-public/jesc-configurator #13: JESC Bricked my ESC

Exentrik1, 2019-12-13:
I initially flashed the 48khz firmware to my hobbywing 40a 4in1 ESC. Then i realized it wasnt working. i then purchased the license for each of the ESC's. Upon power cycling, ESC is not being picked up. I've tried everything including old BLHELI suite to try and force a firmware flash. But now its saying that ESC 2 either has a bad data connection or no power to it. My quad was flying fine before all of this!!

joelucid, 2019-12-14:
@Exentrik1 - sorry, what a bummer! Could you please upload the jesc-configurator debug log so I can check what went wrong? The [Troubleshooting Guide](https://jflight.net/index.php?route=information/information&language=en-gb&information_id=9) also has an item on these issues but I'd like to diagnose  your logs.

Exentrik1, 2019-12-16:
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Installed Configurator Version: 1.2.6, latest version: 1.2.6
Installed Configurator Version: 1.2.6, latest version: 1.2.6
Last used port wasn't saved "yet", auto-select disabled.
PortHandler - Found: COM3
Connecting to: COM3
SERIAL: Connection opened with ID: 1, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 1
MultiWii API version <span style="color: green">received</span> - <strong>1.43.0</strong>
Flight controller info, identifier: <strong>EMUF</strong>, version: <strong>0.1.37</strong>
Running firmware released on: <strong>Dec 11 2019 11:29:35</strong>
Board: <strong>ROSD</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x2b0039344d501020323447</strong>
Reading setup
sending cmd_ProtocolGetVersion 0 0
received cmd_ProtocolGetVersion ACK_OK 0 108
Protocol Version: 108
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead b0 4
received cmd_DeviceRead ACK_OK b0 0,195,232,149
sending cmd_DeviceRead fbfc 4
received cmd_DeviceRead ACK_OK fbfc 34,144,165,255
sending cmd_DeviceRead 3e00 5
received cmd_DeviceRead ACK_OK 3e00 2,251,192,143,130
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,7,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,2,255,255,255,3,251,40,80,4,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,50,53,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceRead ffc0 16
received cmd_DeviceRead ACK_OK ffc0 58,61,101,202,67,89,231,17,173,55,132,135,83,226,169,142
uid: 3a3d65ca4359e711ad37848753e2a98e
sending cmd_DeviceRead 250 128
received cmd_DeviceRead ACK_OK 250 251,32,53,9,210,53,65,93,48,53,2,194,53,32,57,2,178,53,195,234,51,250,235,51,251,235,43,59,59,249,195,235,19,251,234,19,250,234,41,250,235,52,0,251,48,227,4,122,255,123,255,229,37,84,3,96,27,32,42,24,229,93,51,249,235,112,7,195,233,154,64,2,233,250,229,60,196,51,42,250,235,52,0,251,195,234,51,196,84,15,248,235,51,196,84,240,72,248,112,5,234,96,2,120,1,144,0,0,210,168,210,170,229,34,96,2,21,34,129,212,144,0,0,210,168,210,170,97
sending cmd_DeviceRead 2d0 128
received cmd_DeviceRead ACK_OK 2d0 231,192,208,192,224,194,207,5,65,229,102,96,9,229,101,96,5,117,101,0,97,4,117,101,1,229,35,96,2,21,35,195,229,84,96,5,117,85,0,97,4,229,85,36,1,245,85,80,3,117,85,255,208,224,208,208,50,194,175,83,230,127,117,146,250,117,147,255,194,32,83,145,127,210,175,50,192,224,229,138,48,52,8,133,107,139,240,163,208,224,50,208,224,194,175,83,230,239,192,208,192,224,192,240,210,211,210,175,83,136,239,168,138,169,140,170,64,48,141,1,10,117,138,0,117,140,0,117
sending cmd_DeviceRead 350 128
received cmd_DeviceRead ACK_OK 350 64,0,67,136,16,229,102,112,10,195,232,51,248,233,51,249,234,51,250,48,51,31,232,196,84,15,250,233,196,84,240,74,250,195,19,42,250,195,232,51,248,233,51,249,232,42,250,233,52,6,251,97,216,48,50,15,195,232,51,233,51,250,232,42,250,233,52,0,251,97,216,48,49,21,232,117,240,86,164,170,240,233,117,240,86,164,42,250,197,240,52,0,251,97,216,195,234,19,250,233,19,249,232,19,248,232,117,240,86,164,170,240,233,117,240,86,164,42,250,197,240,52,0,195,19,251,234,19
sending cmd_DeviceRead 3d0 128
received cmd_DeviceRead ACK_OK 3d0 250,195,235,19,251,234,19,250,195,235,148,18,80,9,195,234,148,52,235,148,7,80,22,5,34,229,34,112,2,21,34,195,229,34,148,50,80,2,161,40,117,84,0,161,43,229,34,96,2,21,34,48,54,6,124,0,125,8,129,23,172,49,173,50,48,58,4,172,51,173,52,195,234,156,250,235,157,251,146,8,174,96,175,97,48,58,66,80,7,32,53,9,210,53,129,53,48,53,2,194,53,48,53,4,174,98,175,99,32,57,2,178,53,195,234,51,250,235,51,251,162,8,80,10,234,244,36,1
sending cmd_DeviceRead 450 128
received cmd_DeviceRead ACK_OK 450 250,235,244,52,0,251,195,234,148,80,250,235,148,0,251,80,20,120,0,122,0,123,0,129,117,162,8,80,8,120,0,122,0,123,0,129,212,229,37,84,3,96,27,32,42,24,229,93,51,249,235,112,7,195,233,154,64,2,233,250,229,60,196,51,42,250,235,52,0,251,234,142,240,164,249,170,240,235,142,240,164,42,250,197,240,52,0,251,195,235,19,253,234,19,248,195, [... cut ...] ,235,43,59,59,249,195,235,19,251,234,19,250,234,41,250,235,52,0,251,48,227,4,122,255,123,255,229,37,84,3,96,26,32,42,23,229,88,51,249,235,112,7,195,233,154,64
received cmd_DeviceWrite ACK_OK 200 0
sending cmd_DeviceWrite 300 2,233,250,229,54,196,42,250,235,52,0,251,195,234,51,196,84,15,248,235,51,196,84,240,72,248,112,5,234,96,2,120,1,195,235,19,251,234,19,250,144,0,0,229,34,96,2,21,34,97,146,144,0,0,210,168,210,170,97,124,192,208,192,224,194,207,5,58,229,92,96,5,117,92,0,97,107,117,92,1,229,35,

joelucid, 2019-12-17:
@Exentrik1 unfortunately only part of the log made it into the issue. It's better if you drag & drop the log in so it gets attached as file.

Exentrik1, 2019-12-18:
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Installed Configurator Version: 1.2.6, latest version: 1.2.6
Installed Configurator Version: 1.2.6, latest version: 1.2.6
Last used port wasn't saved "yet", auto-select disabled.
PortHandler - Found: COM3
Connecting to: COM3
SERIAL: Connection opened with ID: 1, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 1
MultiWii API version <span style="color: green">received</span> - <strong>1.43.0</strong>
Flight controller info, identifier: <strong>EMUF</strong>, version: <strong>0.1.37</strong>
Running firmware released on: <strong>Dec 11 2019 11:29:35</strong>
Board: <strong>ROSD</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x2b0039344d501020323447</strong>
Reading setup
sending cmd_ProtocolGetVersion 0 0
received cmd_ProtocolGetVersion ACK_OK 0 108
Protocol Version: 108
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead b0 4
received cmd_DeviceRead ACK_OK b0 0,195,232,149
sending cmd_DeviceRead fbfc 4
received cmd_DeviceRead ACK_OK fbfc 34,144,165,255
sending cmd_DeviceRead 3e00 5
received cmd_DeviceRead ACK_OK 3e00 2,251,192,143,130
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,7,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,2,255,255,255,3,251,40,80,4,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,50,53,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceRead ffc0 16
received cmd_DeviceRead ACK_OK ffc0 58,61,101,202,67,89,231,17,173,55,132,135,83,226,169,142
uid: 3a3d65ca4359e711ad37848753e2a98e
sending cmd_DeviceRead 250 128
received cmd_DeviceRead ACK_OK 250 251,32,53,9,210,53,65,93,48,53,2,194,53,32,57,2,178,53,195,234,51,250,235,51,251,235,43,59,59,249,195,235,19,251,234,19,250,234,41,250,235,52,0,251,48,227,4,122,255,123,255,229,37,84,3,96,27,32,42,24,229,93,51,249,235,112,7,195,233,154,64,2,233,250,229,60,196,51,42,250,235,52,0,251,195,234,51,196,84,15,248,235,51,196,84,240,72,248,112,5,234,96,2,120,1,144,0,0,210,168,210,170,229,34,96,2,21,34,129,212,144,0,0,210,168,210,170,97
sending cmd_DeviceRead 2d0 128
received cmd_DeviceRead ACK_OK 2d0 231,192,208,192,224,194,207,5,65,229,102,96,9,229,101,96,5,117,101,0,97,4,117,101,1,229,35,96,2,21,35,195,229,84,96,5,117,85,0,97,4,229,85,36,1,245,85,80,3,117,85,255,208,224,208,208,50,194,175,83,230,127,117,146,250,117,147,255,194,32,83,145,127,210,175,50,192,224,229,138,48,52,8,133,107,139,240,163,208,224,50,208,224,194,175,83,230,239,192,208,192,224,192,240,210,211,210,175,83,136,239,168,138,169,140,170,64,48,141,1,10,117,138,0,117,140,0,117
sending cmd_DeviceRead 350 128
received cmd_DeviceRead ACK_OK 350 64,0,67,136,16,229,102,112,10,195,232,51,248,233,51,249,234,51,250,48,51,31,232,196,84,15,250,233,196,84,240,74,250,195,19,42,250,195,232,51,248,233,51,249,232,42,250,233,52,6,251,97,216,48,50,15,195,232,51,233,51,250,232,42,250,233,52,0,251,97,216,48,49,21,232,117,240,86,164,170,240,233,117,240,86,164,42,250,197,240,52,0,251,97,216,195,234,19,250,233,19,249,232,19,248,232,117,240,86,164,170,240,233,117,240,86,164,42,250,197,240,52,0,195,19,251,234,19
sending cmd_DeviceRead 3d0 128
received cmd_DeviceRead ACK_OK 3d0 250,195,235,19,251,234,19,250,195,235,148,18,80,9,195,234,148,52,235,148,7,80,22,5,34,229,34,112,2,21,34,195,229,34,148,50,80,2,161,40,117,84,0,161,43,229,34,96,2,21,34,48,54,6,124,0,125,8,129,23,172,49,173,50,48,58,4,172,51,173,52,195,234,156,250,235,157,251,146,8,174,96,175,97,48,58,66,80,7,32,53,9,210,53,129,53,48,53,2,194,53,48,53,4,174,98,175,99,32,57,2,178,53,195,234,51,250,235,51,251,162,8,80,10,234,244,36,1
sending cmd_DeviceRead 450 128
received cmd_DeviceRead ACK_OK 450 250,235,244,52,0,251,195,234,148,80,250,235,148,0,251,80,20,120,0,122,0,123,0,129,117,162,8,80,8,120,0,122,0,123,0,129,212,229,37,84,3,96,27,32,42,24,229,93,51,249,235,112,7,195,233,154,64,2,233,250,229,60,196,51,42,250,235,52,0,251,234,142,240,164,249,170,240,235,142,240,164,42,250,197,240,52,0,251,195,235,19,253,234,19,248,195, [... cut ...]
received cmd_DeviceWrite ACK_OK 200 0
sending cmd_DeviceWrite 300 2,233,250,229,54,196,42,250,235,52,0,251,195,234,51,196,84,15,248,235,51,196,84,240,72,248,112,5,234,96,2,120,1,195,235,19,251,234,19,250,144,0,0,229,34,96,2,21,34,97,146,144,0,0,210,168,210,170,97,124,192,208,192,224,194,207,5,58,229,92,96,5,117,92,0,97,107,117,92,1,229,35,96,2,21,35,195,229,79,96,5,117,80,0,97,107,229,80,36,1,245,80,80,3,117,80,255,208,224,208,208,50,192,224,229,138,133,97,139,240,163,208,224,50,5,34,229,34,112,2,21,34,195,229

Exentrik1, 2019-12-18:
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Installed Configurator Version: 1.2.6, latest version: 1.2.6
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Installed Configurator Version: 1.2.6, latest version: 1.2.6
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Running - OS: <strong>Windows</strong>, Chrome: <strong>78.0.3904.97</strong>, Configurator: <strong>1.2.6</strong>
Installed Configurator Version: 1.2.6, latest version: 1.2.6
Installed Configurator Version: 1.2.6, latest version: 1.2.6
Last used port wasn't saved "yet", auto-select disabled.
PortHandler - Found: COM3
Connecting to: COM3
SERIAL: Connection opened with ID: 1, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 1
MultiWii API version <span style="color: green">received</span> - <strong>1.43.0</strong>
Flight controller info, identifier: <strong>EMUF</strong>, version: <strong>0.1.37</strong>
Running firmware released on: <strong>Dec 11 2019 11:29:35</strong>
Board: <strong>ROSD</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x2b0039344d501020323447</strong>
Reading setup
sending cmd_ProtocolGetVersion 0 0
received cmd_ProtocolGetVersion ACK_OK 0 108
Protocol Version: 108
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
sending cmd_DeviceRead b0 4
received cmd_DeviceRead ACK_OK b0 0,195,232,149
sending cmd_DeviceRead fbfc 4
received cmd_DeviceRead ACK_OK fbfc 34,144,165,255
sending cmd_DeviceRead 3e00 5
received cmd_DeviceRead ACK_OK 3e00 2,251,192,143,130
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 16,7,33,255,255,255,255,255,255,9,255,2,255,85,170,1,255,255,255,255,255,2,255,255,255,3,251,40,80,4,255,2,255,122,255,7,1,255,255,0,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,65,95,72,95,50,53,35,32,32,32,32,32,32,32,32,35,66,76,72,69,76,73,36,69,70,77,56,66,50,49,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceRead ffc0 16
received cmd_DeviceRead ACK_OK ffc0 58,61,101,202,67,89,231,17,173,55,132,135,83,226,169,142
uid: 3a3d65ca4359e711ad37848753e2a98e
sending cmd_DeviceRead 250 128
received cmd_DeviceRead ACK_OK 250 251,32,53,9,210,53,65,93,48,53,2,194,53,32,57,2,178,53,195,234,51,250,235,51,251,235,43,59,59,249,195,235,19,251,234,19,250,234,41,250,235,52,0,251,48,227,4,122,255,123,255,229,37,84,3,96,27,32,42,24,229,93,51,249,235,112,7,195,233,154,64,2,233,250,229,60,196,51,42,250,235,52,0,251,195,234,51,196,84,15,248,235,51,196,84,240,72,248,112,5,234,96,2,120,1,144,0,0,210,168,210,170,229,34,96,2,21,34,129,212,144,0,0,210,168,210,170,97
sending cmd_DeviceRead 2d0 128
received cmd_DeviceRead ACK_OK 2d0 231,192,208,192,224,194,207,5,65,229,102,96,9,229,101,96,5,117,101,0,97,4,117,101,1,229,35,96,2,21,35,195,229,84,96,5,117,85,0,97,4,229,85,36,1,245,85,80,3,117,85,255,208,224,208,208,50,194,175,83,230,127,117,146,250,117,147,255,194,32,83,145,127,210,175,50,192,224,229,138,48,52,8,133,107,139,240,163,208,224,50,208,224,194,175,83,230,239,192,208,192,224,192,240,210,211,210,175,83,136,239,168,138,169,140,170,64,48,141,1,10,117,138,0,117,140,0,117
sending cmd_DeviceRead 350 128
received cmd_DeviceRead ACK_OK 350 64,0,67,136,16,229,102,112,10,195,232,51,248,233,51,249,234,51,250,48,51,31,232,196,84,15,250,233,196,84,240,74,250,195,19,42,250,195,232,51,248,233,51,249,232,42,250,233,52,6,251,97,216,48,50,15,195,232,51,233,51,250,232,42,250,233,52,0,251,97,216,48,49,21,232,117,240,86,164,170,240,233,117,240,86,164,42,250,197,240,52,0,251,97,216,195,234,19,250,233,19,249,232,19,248,232,117,240,86,164,170,240,233,117,240,86,164,42,250,197,240,52,0,195,19,251,234,19
sending cmd_DeviceRead 3d0 128
[... cut ...]
sending cmd_DeviceWrite 200 234,51,250,5,130,224,252,195,157,195,152,64,197,152,234,51,250,5,130,224,253,195,156,195,152,64,183,152,234,51,250,5,130,224,252,195,157,195,152,64,169,152,234,51,250,5,130,224,253,195,156,195,152,64,155,152,234,51,250,5,130,224,252,195,157,195,152,64,141,152,234,51,250,5,130,235,196,107,106,84,240,249,234,196,48,51,1,244,84,240,195,153,96,9,144,0,0,210,168,210,170,97,124,210,192,235,244,196,84,240,249,234,244,196,84,15,73,250,235,244,196,84,15,251,195,234,249,148,96,250,235,148,0,251,80,32,195,233,123,0,122,0,121,0,96,22,195,19,80,6,249,195,149,77,96,10,233,245,77,117,78,0,121,0,128,2,5,78,48,58,39,195,234

joelucid, 2019-12-30:
@Exentrik1 - Still the log doesn't include any flash errors. If you drag and drop the file into the browser while editing the github issue it should be attached as file, not included in the comment. That way all the contents will be visible to me.