jflight-public/jesc-configurator #15: brick esc

onlybattery, 2019-12-13:
recognizes one esc with automatic recognition, can do same kind of operation with other two , but not automatically recognize, the fourth esc is separeted, no telemetry window , no update all window. from a flayng quadcopter i have got a non-flyimg one for money .esc are spedix es 30 , bl heli recognize like ch25

onlybattery, 2019-12-14:
[kautefcLog.txt](https://github.com/jflight-public/jesc-configurator/files/3963196/kautefcLog.txt)

joelucid, 2019-12-14:
Hey @onlybattery - just looked at your log. Multiple write operations on the escs failed but all escs still look operational. Please check the [Troubleshooting guide - item 2](https://jflight.net/index.php?route=information/information&language=en-gb&information_id=9).

onlybattery, 2019-12-26:
I tried everything from the list of problem solutions. Nothing change.. One works, two do not connect automatically, but can do something with them, the name of the upgrade is not ok, which means some deviations. And finally one is dead by definition,
  I changed esc connections and even changed positions, nothing changes

 

 

----- Reply to message -----
Subject: Re: [jflight-public/jesc-configurator] brick esc (#15)
Date: sestd., 14 dec. 2019, 12:04

joelucid, 2019-12-30:
Hmm - can blheli-configurator flash these escs without problems? If not it might be a defective esc. Btw even if the flash of an esc becomes corrupted during flashing you can still flash it via the c2 pads. JB has a video on it: https://youtu.be/b-7VFt3IhhE

onlybattery, 2019-12-30:
After trying to upgrade with jflight also bl heli doesn't automatically recognize two and one esc has died while doing the upgrade. I flew this quadcopter before, with bf 3.5 and standard fw everything worked ok. Listened to the miracle stories of how better the rpm filter is and here's a look at the problems with esc improvements.
I understand shit happens, I'll try after your suggestion from JB video. Thank you

 

 

----- Reply to message -----
Subject: Re: [jflight-public/jesc-configurator] brick esc (#15)
Date: pirmd., 30 dec. 2020, 08:54

onlybattery, 2019-12-31:
I've seen that upgrading one by one esc is safer, better results or is it true?

 

 

----- Reply to message -----
Subject: Re: [jflight-public/jesc-configurator] brick esc (#15)
Date: pirmd., 30 dec. 2020, 08:54

joelucid, 2020-01-06:
The good thing about going one by one is that you can stop when you have problems. For example if the lipo isn't fully charged these things can happen.

joelucid, 2020-02-13:
This is now subsumed under issue https://github.com/jflight-public/jesc-configurator/issues/41

closed 2020-02-13 by joelucid