jflight-public/jesc-configurator #22: Can't arm - RPM Filter error on 96kHz beta on TinyHawk 1s.

szafranski, 2020-01-06:
As mentioned I get rpm filter error on osd and 100% error rating in Betaflight.

Model: TinyHawk 1s

Tried on DSHOT 150,300 and 600 on 4kHz PID loop.

Used JESC 1.2.7. Beta

szafranski, 2020-01-06:
Ok, solved that one - didn;t know that telemetry also has to be updated. 
Right now I have next problem - few motors often don't arm at all. Is it known issue?

joelucid, 2020-01-06:
You mean some motors don't spin up on arming, right? Some people have found they need to increase dshot_idle_value with 96khz pwm. Could you try that out?

szafranski, 2020-01-06:
You mean Motor Idle Throttle Value? I have set it to 10% already...

joelucid, 2020-01-14:
Hmm - have you tried the RC 1 which is now available in jess-configurator?

szafranski, 2020-01-18:
Tried it. Can't arm. In 48 version problem doesn't exist. Cheers

joelucid, 2020-01-24:
I've yesterday seen more of this type of issue while debugging some changes. I found that 48khz pwm is not able to spin a motor at the same low rpm as 24khz. This might be the same problem. If it is it might not be solvable for every motor.

One thing you could do is try out different startup power setting for your esc.

szafranski, 2020-01-24:
Maybe it is dumb idea but can pwm freq be changed after arming? Maybe after eg 3 secs it could be changed from 24 to 48 etc

joelucid, 2020-01-25:
Yeah - that or a full open loop startup - eg sinusoidal. For now just stick with 48khz pwm then. The savings from 48 to 96 are relatively small.

closed 2020-01-25 by joelucid