jflight-public/jesc-configurator #23: ESC Missing after flashing JESC

speedylolo, 2020-01-07:
Hi Joe, I tried Jesc configurator and tried to flash tinyhawk freestyle’s  esc with it. I don’t really know what’s happened. But unfortunately now one esc is missing. I tried every thing describ on the troubleshooting and nothing work. I tried also to get it back by using BLheli suite.

[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4030675/Log.txt)
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4030680/Log.txt)

speedylolo, 2020-01-08:
Are you gonna help me?

joelucid, 2020-01-14:
Hey @speedylolo - just looked at your log. First there was a failed attempt to write settings while flashing esc 4. Then later reads to other escs failed as well. I can only imagine that either your usb cable connection wasn't reliable or there is a defect with the board.

If the esc doesn't come back you can try flashing using the c2 pads on the esc. JB has a video on it: https://www.youtube.com/watch?v=b-7VFt3IhhE.

speedylolo, 2020-01-16:
Thank you it came backdoor!

closed 2020-01-24 by joelucid