jflight-public/jesc-configurator #24: [Resolved] ESC Missing after failed JESC flash using JESC Configurator and CPU Overclocked

MRC3742, 2020-01-11:
I am posting this as an information post and possibly to get a warning added to the JESC Install Instructions. NOTE: DO NOT FLASH ESC's WITH CPU OVERCLOCKED!

When flashing DYS ELF running BF Performance Edition 4.11 EA3 SPRacingF3 and 4in1 ESC using A-H-70 version 2.3-48mhz firmware, ESC#1, ESC#3, and ESC#4 "Flash Failed".

This only happened after I had applied the licenses and was flashing the new firmware to activate the licenses. I then tried to reflash same firmware with same results. I also tried to flash version 2.2-24mhz but also failed flash reported.

I had read back in an RC Groups post about someone having trouble flashing with their CPU overclocked. I was running the F3 CPU at 120MHZ when attempting the original flash. I entered the CLI and set CPU_Overclock = OFF. I then went back to JESC Configurator and now ESC#1 was missing? I could now flash the other three ESC's to version 2.3-48mhz and all flashed properly and were activated. So this seems to be related to CPU overclock.

Trying multiple reads on JESC, BLHeli Congurator or BLHeli Suite with no success to read ESC#1. Possibly corrupted the bootloader on the first attempt on that ESC?
Searched and found someone that had fixed a similar missing 4in1 ESC issue on a different board flashing just a single ESC using the pads on board with BLHeli Suite and an Arduino Nano using the C2 mode.
https://www.rcgroups.com/forums/showpost.php?p=36908650&postcount=627

After researching the pinout of the Silabs EFM8BB21F16 I was able to locate the on board pads for C2D and C2CK and then successfully flashed BLHeli 16.7 to ESC#1. It would now show up when using the JESC Configurator and eventually was able to successfully update it to Version  2.3-48mhz and install all the telemetry firmware. After successful CPU flashing, overclocking the CPU is again necessary to be able to run the F3 with Bidirectional DSHOT 300, 4k/2K, RPM Filtering and Dynamic Notch Filter.

Not sure if this is related to the other issues listed as missing ESC but could possibly be linked to overclocking the CPU when trying to update to licensed JESC firmware?

PLEASE NOTE: DO NOT FLASH ESC's WITH CPU OVERCLOCKED! - Rich

joelucid, 2020-01-14:
Thanks! This is good feedback. Generally I have found 120mhz to be too borderline to be useful. I'd go one step slower.

MRC3742, 2020-01-14:
Yes, I agree and am currently running CPU_Overclock = 112MHZ
As you have posted 128MHZ is a no go and locks up the board requiring a complete board reflash with chip erase to recover. 120MHZ works fine but 112MHZ must be safer?

I had flashed on the same JESC version to these ESC's earlier with no issue - but without a license. (Just to make sure these ESC's were flashable) After I added the license, then all 4 were listed as licensed - but not activated. When I reflashed the current hex to activate them the progress bar slowed (stopped) about halfway and I think that's where the activation was being initialized and the cause of the issue? But definitely due to the overclock as the root cause.

Now before flashing I first drop the Gyro/ACC to 1K/1K then drop the overclock to OFF. (I'm running 4K/2K and if I just drop the overclock then CPU is running at 100% - Haven't tried flashing at 100% but can't be good)

I saw a small increase with 48khz firmware but currently running 2.3 96khz A-H-70 with great results. Motors are much quieter and flight time up about 25% from 4 minutes and adds an additional 45 seconds. Running 2S 600Mah Batts

Thanks - Rich

cbudd75, 2020-04-04:
> Silabs EFM8BB21F16-QFN20 datasheet

would you be able to figure out what  i need to solder to on an BetaFPV 1s v2.1 to reflash my 1 esc that is no longer showing up?

JonRascol, 2020-04-05:
Thats what i need to know too

joelucid, 2020-04-05:
[image]

Unfortunately this board doesn't have the c2 pads that make flashing via c2 easy. To flash you need C2D, C2CK and GND. GND is easy. You could try to solder to the chip directly for C2D and C2CK but it'll be a bit painful.

[image]

JonRascol, 2020-04-05:
Ah so basically it’s bricked?

[Screen Shot 2020-04-05 at 6 13 07 PM]<https://user-images.githubusercontent.com/18134492/78503797-238dfa00-7769-11ea-9878-497a6980e12f.png>

[Screen Shot 2020-04-05 at 6 15 15 PM]<https://user-images.githubusercontent.com/18134492/78503873-9dbe7e80-7769-11ea-84c3-66d1101cbcb5.png>