jflight-public/jesc-configurator #25: ESC Missing After Flash

KWX101, 2020-01-18:
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4080121/Log.txt)

I hope this shows you what you are looking for.... I was one of those guys that didnt read all of the documents and tried to fix it.

Flashed JESC_OL5_24_2_3.HEX then went into betaflight(didn't see the failed flash right away) and motor 2 didnt spin.  went back to JESC config to try to recover and flashed back to factory.  

Let me know if you need any more info.  I know enough to be dangerous to my equipment(why im here...) but may ask novice questions.

Thanks!
Jake

mat277, 2020-01-21:
have you solve your problem?

KWX101, 2020-01-22:
Nope...  no matter what i do it still will not recognize esc#2.

KWX101, 2020-01-22:
@joelucid ...?

joelucid, 2020-01-24:
This log shows that jesc-configurator could not read the settings of this esc before even trying to flash it:

`sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,100,1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,100,1
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 177,232,100,1
`

Did you flash using blhelis-configurator before using jesc-configurator?

KWX101, 2020-01-24:
I did not.  I flashed jesc then tried to flash back to blheli-s after I noticed #2 wasn't working.  The log that I sent you was yes after I tried blheli-s again.  A friend of mine lost 2 escs on another meteor65, that quad had h type esc.
[image]
[image]

joelucid, 2020-01-24:
Ah ok. The state that you show here with *flash failed* in jesc-configurator is recoverable (see troubleshooting guide on jflight.net). But the log you sent shows an esc that can't be read at all.

Now given that you had problems flashing it with JESC-configurator it's possible that the flash using blheli-configurator corrupted the flash memory of the esc or maybe there's an intermittent problem (sometimes escs come back with fresh battery etc).

There's been one report that flashing can fail using overclocked f3 - were you running that?

KWX101, 2020-01-24:
Not overclocked, just a standard week old meteor65... any way to recover the esc that your aware of?

KWX101, 2020-01-24:
I couldn't get it to show up in blheli suite either.

joelucid, 2020-01-25:
Yes, you can flash it using the c2 interface. JB has a video on it: https://www.youtube.com/watch?v=b-7VFt3IhhE

KWX101, 2020-01-26:
I wonder what pads if any are available on the meteor's FC to do that though...

KWX101, 2020-01-28:
so,  after some digging.... the ic2 pads are not broken out on a betafpv v2.1.   so now what?

joelucid, 2020-01-28:
I've circled the c2 pads on this image of the betafpv v2.1 for your convenience:
[image]

KWX101, 2020-01-28:
Thanks Joe but that is actually the V2.0.... just a moment I'll upload a pic...

KWX101, 2020-01-28:
Well I stand corrected... that may be the 2.1, but I have the 2.2 anyhow...
[image]
[image]

KWX101, 2020-01-28:
Well crap... this is a replacement for the "bad" one. maybe the one that is "bad" has the V2.X.  lol I dont know anymore... smh.  I'll take a closer look when I get back across the state tomorrow.

joelucid, 2020-01-28:
Usually they all have the pads since the boards need to be flashed post reflow (flash may be corrupted during reflow).

joelucid, 2020-01-28:
It would be on the other side where the efm8 chips are located.

KWX101, 2020-01-28:
Right, there is supposed to be 2 pics in that post.  
[image]

joelucid, 2020-01-28:
Indeed that doesn't look so good :(. I suppose you could try to solder to the chip directly - but that does get a bit finicky.

joelucid, 2020-02-13:
Sometimes when I never see a problem in the log the problem is due to configuration issues in BF. I've created a summary issue for these problems: https://github.com/jflight-public/jesc-configurator/issues/42.

The screenshots you sent just show "flash failed". That can be handled according to this issue: https://github.com/jflight-public/jesc-configurator/issues/40

C2 flashing is only required when the esc has corrupted flash and none of the safeguards worked.

cbudd75, 2020-04-04:
> Indeed that doesn't look so good :(. I suppose you could try to solder to the chip directly - but that does get a bit finicky.

Is there away to figure out what part of the chip i need to solder too?  I have this same problem.

Landairfpv, 2020-05-10:
Same problem. Betaflight F4 1s fc V2.2. I only got two flights in with the stock BLheli 16.6 that came on the AIO. I wanted to see what the flight time difference would be using JESC firmware. Well I found out... Has anyone found a solution to this. I had motors remapped and ESC#1 was lost but actually will be esc.

automatt1c, 2020-07-06:
So how do you actually flash using those pad on the betafpv v2.0?  Is one pad the c2d and the other the c2ck? how do you know which pad is which?

kmeda, 2020-08-07:
> I did not. I flashed jesc then tried to flash back to blheli-s after I noticed #2 wasn't working. The log that I sent you was yes after I tried blheli-s again. A friend of mine lost 2 escs on another meteor65, that quad had h type esc.
> [image]
> [image]

I have the same issue on my very first tiny whoop build. ESC#4 wont show up :(
My LiPo must have drained out on esc 4 when I flashed blheli_s back again.

evolvewd, 2021-06-09:
same issue here, esc 3 lost due to flash fail, any suggest?
[image]

automatt1c, 2021-06-09:
> same issue here, esc 3 lost due to flash fail, any suggest?
> [image]

The only Fix is to flash with the c2 connections. It's usually a ridiculously small pad on the near the ESC chip.  if i remember right there might be a c2 pad for each esc on the AIO board. 

Some boards don't have C2 pads and then the AIO is junk. 

This video shows you the process. you need a flasher, i used an arduino nano. 
https://www.youtube.com/watch?v=b-7VFt3IhhE

evolvewd, 2021-06-09:
i've seen the video, thanks for your answer, but on the 2.2 board i can't find the pins.. have you flashed a 2.2?

automatt1c, 2021-06-09:
Mine is a v2.0 no RX module. I think only the no RX boards have c2 pads. i actually lifted the c2 pad when i tried to solder a wire but was able to  scratch the PCB to get to the trace.  That's a good idea to flash the chips directly but sounds like a bit of a nightmare to get a wire to those pins.

joelucid, 2021-06-10:
Yeah if you have steady hands and young eyes or a microscope you can try soldering to the chips. It's unfortunate that some boards don't have the c2 pads. These used to be standard. Normally the escs are flashed after reflowing the board so the pads are needed. But I guess for some of these boards without pads the efm8 chips are flashed prior to reflow. I've heard that manufacturers generally like to avoid this approach since reflowing can sometimes affect flash memory but clearly not every manufacturer does this. The push for miniaturization might be behind taking a bit of risk on yield.

joelucid, 2021-06-10:
If you look at the data sheet https://www.silabs.com/documents/public/data-sheets/efm8bb2-datasheet.pdf page 42 ff you can see the pinout of the chip. C2CK and C2D comprise the C2 interface.

evolvewd, 2021-06-10:
thank you yes i found the component datasheet and i started soldering a wire then i had to go out for work i will continue later. however, I keep you updated if it should work at least some instructions will remain written for the next unfortunate
[image]

evolvewd, 2021-06-10:
Fixed! 
Flashed to blheli stock through Arduino Uno e 4wire connection than re-flashed to JSC.
All works fine 
:)
[image]

reednoel4u, 2021-08-23:
Hi All,
My FC ver 2.2 breakout rx-spi.
please show me, where point that, I can wiring rx-sbus external ( ie XM+, Lr1000).
Thanks so muchs.