jflight-public/jesc-configurator #26: lost a esc

mat277, 2020-01-21:
Hello!  After flashing with JESC i lost esc 3

Mobula 7 F3 FC

Can you help me?

[Mat277Log.txt](https://github.com/jflight-public/jesc-configurator/files/4088573/Mat277Log.txt)

joelucid, 2020-01-24:
Your esc firmware was flashed incorrectly since during flashing the connection to the esc was interrupted. This could be due to an unreliable usb connection or insufficient power. The BLHeli-S/jesc flashing procedure uses a number of strategies to avoid this from happening but it can't provide 100% protection.

You can always fix this by flashing using the C2 pads on the esc.

closed 2020-01-24 by joelucid

mat277, 2020-01-25:
I check the video, and i have no clue how to do it on a small board like that!  I think i need special tool to do that to ?!