jflight-public/jesc-configurator #27: lost 3 ESC after some flashing

ThLehmann, 2020-01-21:
hello,

just want to update my O-H-5 ESC from a BetaFPV meteor65 to JESC. After flashing to V2.2 i could'nt licence the ESC's because the button stays unaccessable. So i tryed out the other ones, V2.3 for H version, back to V2.2 48/24kHz. The licence button does'nt came activ.
So i wanna check my new copter without RPM filtering .... But now only one motor is activ, hmm.
Back to JESC and BLHeliSuite i noticed that only ESC #3 is available, all other ones are lost :(
Don'nt know what happend
Hope you can give a solution for this behaviour.
Thomas
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4093934/Log.txt)

ThLehmann, 2020-01-21:
[image]

ThLehmann, 2020-01-21:
[image]

joelucid, 2020-01-24:
Your log only shows the failed attempts to communicated with the esc, no flashing. Can you send the full debug log, please?

ThLehmann, 2020-01-24:
I thought that it was the full log. What else i should do to get the full log ? Sorry for my unknownless

joelucid, 2020-01-24:
Just using the "Save Debug Log" button should do the trick. You need to do it on the machine you used for flashing.

ThLehmann, 2020-01-24:
Ok, made three Logs
after connecting (the one you need?)
after reading setup
after flashing ESC 2+4
[LogAfterConnect.txt](https://github.com/jflight-public/jesc-configurator/files/4107373/LogAfterConnect.txt)
[LogAfterReadSetup.txt](https://github.com/jflight-public/jesc-configurator/files/4107372/LogAfterReadSetup.txt)
[LogAfterFlashing2-4.txt](https://github.com/jflight-public/jesc-configurator/files/4107371/LogAfterFlashing2-4.txt)

joelucid, 2020-01-25:
All of these logs already have at least two escs that don't get detected. The log doesn't get deleted by jesc-configurator so any debug log you save on a machine should contain everything you've done on that machine.

Have you ever seen all escs in configurator? If so can you send me the log from the machine on which you did the initial flash?

ThLehmann, 2020-01-25:
the log is from that machine, and yes all four ESC were seen and could be flashed. I only go on with flashing some other versions because the licence button was not coming available. See my entry post.
The error happens as my battery goes empty i think.
before i open this issue i didnt save any log, the log in first thread was the first one on this machine.
So i am wondering why the log is/should be empty.

joelucid, 2020-02-13:
YEah - no clue really. All I can do is refer you to https://github.com/jflight-public/jesc-configurator/issues/41 for a pointer to how to flash escs using the c2 interface.

closed 2020-02-13 by joelucid