jflight-public/jesc-configurator #30: Issues flashing Diatone Mamba Toothpick AIO G-H-30 ALL ESCs dead except 3 now

PKVillager, 2020-01-28:
Flashed JESC 2.3 48 with RPM for whoop, licensed then noticed ESC 1 showed flash failed. Reflashed and it seemed good. Then flashing telemetry for all failed and ESCs stopped showing up randomly and after reconnecting and rebooting a few times now only ESC 3 shows up. In BF motor tab I cant spin any motors. I can only hear the JESC startup tone very faintly so i think only one is partially booting up. This was on a brand new FC just flashed latest BF and went straight to JESC.
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4120398/Log.txt)

I had no issues with a PYRODRONE AIO-412T previously.

joelucid, 2020-02-04:
The log shows several instances of communication error when accessing the ESCs. I would recommend to try this again with a fully charged lipo, with a USB cable that works well and make sure that the usb connector is also working right.

It does not look like any of the escs were corrupted during your flash attempts. So with the above they should show again in jesc-configurator and you should be able to flash again.

PKVillager, 2020-02-07:
Since I knew my USB was good and I'm sure my 2S pack was fully charged I soldered up an XT60-XT30 so I could feed the board with a 4S pack and I was able to re-flash successfully! THANK YOU!!!

closed 2020-02-07 by PKVillager