jflight-public/jesc-configurator #32: BETAFPV 75x JESC 2.3 48k Problems

grecomafioso, 2020-01-30:
When I flash the JESC 2.3 Version with 48k and telemetry to my BETAFPV 75x, I can fly it for like 5 seconds and then it falls to the ground and spins around it’s Yaw axis with little throttle. If I raise the throttle stick the motors do not spin any faster. The Lipo is fully charged.

With 24k and telemetry the whoop flyes normal.

What’s the problem here?

joelucid, 2020-02-06:
Do the escs get hot when this happens? It could be that betafpv uses a marginal dead-time and applying it twice as often causes the issue at 48khz. In that case you might be able to fix it by using a higher dead-time.

grecomafioso, 2020-02-06:
How can I do that? Need some steps here...

Thanks!

joelucid, 2020-02-12:
I've seen someone else report success in such a case by doubling dead-time. Dead-time is the number in the layout tag. So say your layout is F-H-40 - then you could select F-H-90 instead when flashing. Remember to check the checkbox "ignore incompatible layout" when flashing.

marlic-741, 2020-05-12:
I updated also the beta75x to JESC 2.3 48kHz.
I did an update of betaflight first:
# Betaflight / STM32F405 (S405) 4.1.5 Mar 16 2020 / 05:19:58 (d4e74e39c) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

Betaflight is configured to DHOT600, 8k/8k and runs with 13-16% load

I am running "JESC-PH10-48 (2.3)" without telemetrie.
I will do a testflight tomorrow and could provide some additional information if that would help.

marlic-741, 2020-05-13:
So my testing was successfull. 
Two flights with a 350mAh LiPo 02:45 (to 6.9V) and 02:22 (7.1V).
I had no problems during the flight. I tested LOS some flips and rolls, some FPV.

closed 2020-06-20 by joelucid