jflight-public/jesc-configurator #34: Tried to flash JESC and lost 2 of 4 ESC

arierep60, 2020-02-05:
Hello, I have a Emax Tinyhawk and tried to flash JESC on it to get the 48KHz benefits and now I lost 2 escs :(
Now they show up as ESC 2: ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ (16.77), ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ and I can't flash anything on them, someone could help me please??

joelucid, 2020-02-06:
Hi there - please update to jess-configurator 1.2.8 and read through the instructions in the troubleshooting guide on jflight.net. If you could also send the jesc-configurator debug log I would be very grateful.

arierep60, 2020-02-06:
Hello Joelucid, that's the version I have, 1.2.8, the problem is that if I try to flash those 2 esc that appear with the ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ  nothing happens :(
I will get the log and send you later
thanks

arierep60, 2020-02-06:
here you have the log:
https://pastebin.com/bWD1ZfXP

joelucid, 2020-02-12:
Please read the troubleshooting guide on jflight.net to fix this issue. It has an item on the problem.

arierep60, 2020-02-12:
@joelucid are you sure? I already read that and didn't find any fix for this problem :(

joelucid, 2020-02-12:
Have you checked the "ignore incompatible layout" checkbox? The log you pasted shows that the flash fails because the layout is incompatible.

arierep60, 2020-02-12:
I don't think so, I will try it in a few hours as soon as i get home, but I don't remember to see that checkbox :|

joelucid, 2020-02-13:
https://github.com/jflight-public/jesc-configurator/issues/40

closed 2020-02-13 by joelucid

arierep60, 2020-02-13:
@joelucid tried that and it flashed all the escs!
just have to test flight the drone but I think everything is fine now
thank you so much my friend! ;)