jflight-public/jesc-configurator #35: Meteor65 ESC O-H-5 can't license

ThLehmann, 2020-02-07:
Hello,

have a problem to license the ESC to get the RPM activ. The CPU is of typ BB21F16G C011Q5 1945 C
The L experimental is working, there i got the RPM signaled.
What is the problem ?
Greets
Thomas
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4172810/Log.txt)
[image]

tmon55, 2020-02-09:
I have the same problem with my Meteor65.  However, I flashed to O-L-5 and I have the JESC OL5 48 version 2.3.  Not sure if H or L is the right one.  I can't license or flash telemetry and it doesn't arm.  Please let me know if you find a solution.

tmon55, 2020-02-09:
I have a similar problem with my Meteor 65.  I flashed the OL5-48(2.3 version).  After I do that I can't license or flash telemetry.  Because I can't flash telemetry my Meteor 65 doesn't work.  Did I flash the right firmware?  How do I flash telemetry and/or license the ESC's?

joelucid, 2020-02-12:
For L escs you don't need to license anything for telemetry. Just flash the "rpm preview" version. It supports rpm telemetry out of the box. The normal L 2.3 does not support rpm telemetry yet.

joelucid, 2020-02-12:
Oh sorry - @ThLehmann your problem is different. You likely have a mistagged fw which claims to use a L chip. I have a fix for this in beta here: https://jflight.net/downloads/JESC_CONFIGURATOR_BETA/

ThLehmann, 2020-02-12:
@joelucid, pefect work, now it is in functionality, could license and activate my O-H-5 ESC
Thank yo very much for spending your time for us!
Danke

tmon55, 2020-02-13:
I changed to the "RPM filtering" version and the "flash telemetry" button is grayed out still.

joelucid, 2020-02-13:
You don’t need to licensed and can’t license L escs. The rpm preview already includes the Telemetry capability. Use dshot 150 and 4k2k looptime.

tmon55, 2020-02-13:
Oops.  Got it.  Thanks.

OperatorB, 2020-03-24:
I got somewhat similar issue I hope you can help. I have 4 ESC came with A-H-25 BLHELI_S fw and I decided to try JESC fw with telemetry. Currently I can flash them with 1-wire interface and so I flashed JESCv2.3 fw.  However, I cannot activate my already purchased license for some reason.
After "ReadSetup" Configurator said "Your ESC is locked...unlock using C2 interface!". Do you have some hint for me? @joelucid

joelucid, 2020-03-27:
@OperatorB - yeah that's a tough one: some vendors lock their escs is a way that prevents jesc from being used. It's very rare so I think these are likely just some batches with a mistake made during flashing. You can make these work by refreshing them using the C2 interface to remove the lock - but it's a bit involved. See https://youtu.be/b-7VFt3IhhE.

OperatorB, 2020-03-27:
I was able to unlock all 4 ESCs with arduino C2 method then I flashed all via 1-wire with the JESC firmware and telemetry. Now they work as I wished for. **Thank you Joe for the help and your work!**
FYI: these are the "Favourite LittleBee.. 30A" ESCs, I think this type is just locked ESC overall.