jflight-public/jesc-configurator #37: Lost esc #1 and additional issues.

bigempty12, 2020-02-13:
This may be long winded, but just trying to figure out what I did wrong. I updated a 3" build that I have with the rpm filtering a couple of weeks ago and all went great. So I decided to update my 5" to try it out. I updated to BF 4.1 and purchased my license for jesc. This is a flying 5" btw. Not a new build. I flew it the day before doing this. During the first flash of the jesc firmware, the flashing failed after esc jflight-public/jesc#1. I then tried again and it completed esc 2,3,and 4. No matter what I tried, I never got esc 1 to be read again, and it never worked again. I tried reading the esc in blheli, with no luck. After approximately an hour of trying to figure out what I might have done, I literally watched my flight controller lights flicker a couple of times, disconnect from my laptop, and never power up again. First question I have is , can these be related? I have had no issues with the esc or flight controller at all. Is it possible that I loaded the incorrect firmware for the esc causing some type of mapping issues that shorted out my flight controller? I know fully that all of this is user beware, I just would like to know what I could have done so that I do not do it again with my new flight controller and esc. Thanks

joelucid, 2020-02-13:
Could you please attach the jesc-configurator debug log? Thank you!

joelucid, 2020-03-05:
Closing due to missing debug log.

closed 2020-03-05 by joelucid