jflight-public/jesc-configurator #42: JESC-configurator can't access some ESCs after BF upgrade

joelucid, 2020-02-13:
Sometimes JESC users first attempt to flash using FC firmware which doesn't support flashing Telemetry. When they upgrade to BF >=4.1 afterwards sometimes the new configuration doesn't allow accessing some of the motors in JESC. This will show as some ESCs visible initially but not after the update.

In this case attempt a fresh install of BF with "full chip erase" to ensure you start from a clean slate.

vspec22001, 2020-02-19:
Is there a fix? I have tried full chip erase and still doesn't work..

joelucid, 2020-02-21:
It really depends on the specific FC. Normally just about all FCs work out of the box after a fresh install. Sometimes there can be the issue that you haven't connected to BF configurator yet so the defaults haven't been installed. After an install always connect with BF cfg first.