jflight-public/jesc-configurator #46: did i brick my esc? is it fixable log #1 partial

donnygfpv, 2020-02-21:
[jesclog1.txt](https://github.com/jflight-public/jesc-configurator/files/4237587/jesclog1.txt)
[jescLog2.txt](https://github.com/jflight-public/jesc-configurator/files/4237589/jescLog2.txt)

ive purchased 20 licenses and have flashed 4 4in1escs with no issues ,went to flash my 5th 4in1 not sure if i bricked it, esc #1 is no longer recognized and only esc 2,3,4 are flashable any help would be greatly appreciated would love to continue to support you and may even by another pack of 20 licenses, just really bummed on throwing away 4 licenses and a perfectly good esc

joelucid, 2020-03-05:
This looks like it's a duplicate to https://github.com/jflight-public/jesc-configurator/issues/43. Two flashes proceed in parallel. You will need to flash using the c2 interface to revive the esc. Along the lines of https://www.youtube.com/watch?v=b-7VFt3IhhE.

cbudd75, 2020-04-04:
I have have the same problem going on with one of my ESCs on my Betafpv v2.1 1s AOI FC.  Is this possible to do on mine as well?

joelucid, 2020-06-20:
@cbudd75 as long as the c2 pads are there you can flash using an arduino, yes.