jflight-public/jesc-configurator #49: Problem flashing ESC's on my Mobula 6

rachoban, 2020-03-06:
I bring up the JESC configurator, then I connect to my Modula 6 via USB, then I plug in a battery, then I select **READ SETUP**,  the ESCs show **BLHELI O-H-5 (16-7)**
The FC is Betaflight 3.5.7

Then I select FLASH ALL, the screen comes up for selecting a target, I select **O-H-5** for esc and **JESC 2.3 48KHz** for the version

Then I have to select just **FLASH**, the tab for FLASH ALL is grayed out and cannot be selected

The problem is nothing happens, nothing is flashed.

JESC 2.3 48KHz is without the RPM filter. I am only doing this to use 48KHz on my Modula 6

If you look at the log attachment I sent, you will see I get some error messages, but I don't know what any of the errors mean or how to fix them. 

[JESC problem Log 3 on 3-5-2020.txt](https://github.com/jflight-public/jesc-configurator/files/4299308/JESC.problem.Log.3.on.3-5-2020.txt)

chenhaus, 2020-03-10:
I figured this one out. If you activate your license on the Mobula6 ESC, then you must update to Betaflight 4.1 otherwise it will fail with the "Version mismatch, check ESC and MCU" error if you try to flash even the basic 48kHz firmware. I'm guessing this was a compatibility check for flashing telemetry for RPM filtering.

I have 2 Mobula6's  on Betaflight 3.5.7 and I ran into the same issue on the first one trying to license, then flash all. The second one, I went straight to flash all, and it worked great.

Will flash Betaflight 4.1 on the first one and see if that allows the licensed ESC to be flashed.

chenhaus, 2020-03-13:
Yup, flashing to Betaflight 4.1 did the trick.

joelucid, 2020-03-27:
@chenhaus is correct - you need BF 4.1 to flash.

closed 2020-03-27 by joelucid