jflight-public/jesc-configurator #51: JESС does not see the ESC

kvognch, 2020-03-25:
Hello!

My configurator connects to the controller, but does not show the ESC. Here is my log:

```
Reading setup
sending cmd_ProtocolGetVersion 0 0
received cmd_ProtocolGetVersion ACK_OK 0 108
Protocol Version: 108
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0
ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
```

How can I fix this? Thanks!

joelucid, 2020-03-27:
This means that the FC could not talk to the esc. Most often the issue is that no power was applied to esc (via battery). If you do have a battery connected check if you can spin the motors in the motors tab. Also make sure bf configurator isn't connected at the same time as jesc configurator.

If fc can spin the motors but jesc configurator still can't read the config please attach the jesc cfg debug log via drag and drop here.

kvognch, 2020-03-28:
The battery is connected. I can spin the motors in the motors tab. BF configurator is disconnected from FC. But nothing changes. Here is my log:
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4397222/Log.txt)
Thank you!

joelucid, 2020-06-20:
In the log I see first a successful flash on a S405 board, but then subsequent read setups fail. Is this the same esc? Or is this a second one which has always had the problem?

kvognch, 2020-06-24:
> In the log I see first a successful flash on a S405 board, but then subsequent read setups fail. Is this the same esc? Or is this a second one which has always had the problem?

The first ESC is another quad; everything is in order with it. And with the second, the problem was constant.

joelucid, 2020-07-02:
Can you see these escs in blheli-configurator? They report a non blheli_s signature:

```
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4
```

kvognch, 2020-07-02:
> Can you see these escs in blheli-configurator? They report a non blheli_s signature:
> 
> ```
> sending cmd_DeviceInitFlash 0 0
> received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4
> ```

That's my fault. My regulator works on Blheli_32. I have carefully read the manual. Thanks for the help and sorry for the trouble!

closed 2020-07-06 by joelucid