jflight-public/jesc-configurator #55: RPM Filter : mix of blheli_32 & blheli_S

asaff73, 2020-04-19:
hi there

i have 3 blheli_s ESCs on my quad and one blheli32 ESC 

 .....  facing problem using RPM filter

purchased and flashed the "S" with that :
[image]

i flashed the "32" with that :
[image]

....and the Betaflight 'refuses' to keep this setting 
[image]

Any ideas how to overcome that ?

tnx!

Asaf

joelucid, 2020-06-20:
That's typically because you pasted incompatible settings. Please start with a clean BF 4.1 - then it should work.

LastRizes, 2020-07-19:
set dshot_burst = ON