jflight-public/jesc-configurator #56: DYS 4in1 30A, Motor 1 not syncing

CjlaBuk, 2020-04-20:
Hello i'm having issues with DYS 4in1 30Amp ESC its a A-H-30 firmware, i'm running 24khz jesc.
Using CLRACINGF7 FC.
No matter what i do my motor 1 always jitters/shutters on arm and then it does bip bip (reboot?) and doesn't spin anymore.
Already Tried:
*Soldering it to another motor output on the FC
*Using another motor
*Tested 4 ESC's (all same brand/model)
*Tested 48/92khz versions
*Tested 80% of startup power values
*Tested every possible value of demag compensation/timing
*have a similar issue on my tinyhawk s when running on 1s battery. 2s works.

when i set set dshot_bitbang=off esc wont work anymore
It may work obsessionally for 1 flight then it fails again without any changes.

Any ideas?

CjlaBuk, 2020-04-20:
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4505750/Log.txt)

joelucid, 2020-06-20:
Have you tried switching off bidirectional dshot to see if it works then?