jflight-public/jesc-configurator #63: 1 esc is missing

ez5, 2020-05-23:
I went through all the troubleshooting and still having problem
 
Any help would be much appreciated

[Log.txt](https://github.com/jflight-public/jesc-configurator/files/4672546/Log.txt)

joelucid, 2020-06-20:
This is interesting: initially you were able to read esc 2 once, but it failed to every other attempt. Consequently jesc-configurator never even attempted to flash it. Could there be a esc wiring issue? Can you spin motor 2?

ez5, 2020-06-20:
It's an AIO Betafpv 1-2s.....No it's not spinning. I put the motors on a different AIO and they work fine

joelucid, 2020-06-21:
Did you ever flash using a different PC? If not it seems the esc just died on you since it no longer responds to commands and was not flashed by jesc-configurator.