jflight-public/jesc-configurator #66: Could not flash one ESC, no licence all or flash all options

MrD-RC, 2020-05-30:
I'm trying to get my ESCs set up with the JESC firmware. When I opened the Configurator, there is not Licence all or Flash all option available, they are greyed out. So I manually flashed each ESC. This worked fine up to ESC 4, which refused to flash. No error was given. I downloaded the hex file from GitHub and was able to flash it with BLHeli suite. But, the configurator still gives no licencing options for the telemetry, nor flash all. ESC 4 also states that it is already licensed?

[image]
[image]

The USB cable is fine, I use it with all my other flight controllers. The LiPo was a 4S with 4.1V per cell. Also, I could flash the firmware using other package.

I have included 2 logs:
**JESC Debug Log - trying to overwrite JESC firmware.txt** This is a log after I had flashed the JESC firmware using BLHeli suite.
**JESC Debug Log - trying to overwrite BLHeli_S firmware.txt** This is a log after I had re-flashed the BLHeli_S firmware back on to the ESC using BLHeli suite. I'm using JESC to try and write the JESC 2.3 24 firmware.
[JESC Debug Log - trying to overwrite JESC firmware.txt](https://github.com/jflight-public/jesc-configurator/files/4705684/JESC.Debug.Log.-.trying.to.overwrite.JESC.firmware.txt)
[JESC Debug Log - trying to overwrite BLHeli_S firmware.txt](https://github.com/jflight-public/jesc-configurator/files/4705685/JESC.Debug.Log.-.trying.to.overwrite.BLHeli_S.firmware.txt)

esinma, 2020-06-13:
I have this same issue after replacing one motor. Tells me 3 esc's are licensed but wont give me the opportunity to flash 4th. Thanks

joelucid, 2020-06-20:
Could you try with these bits, please: https://jflight.net/downloads/jesc-configurator-signature-workaround/

joelucid, 2020-06-21:
You also try the new jesc configurator 1.2.9 beta 1. I included the fix.