jflight-public/jesc-configurator #73: if jflight.net is not reachable from client system, jesc configurator 1.2.8 or 1.2.9 beta 3 cannot fully load

jgauruder, 2020-08-17:
if jflight.net is not reachable from client system, jesc configurator 1.2.8 or 1.2.9 beta 3 cannot fully load

2020-08-16 @ 20:16:31 -- Running - OS: MacOS, Chrome: 78.0.3904.97, Configurator: 1.2.9

2020-08-16 @ 20:16:34 -- Serial port successfully opened with ID: 1

2020-08-16 @ 20:16:34 -- MultiWii API version received - 1.42.0

2020-08-16 @ 20:16:34 -- Flight controller info, identifier: BTFL, version: 4.1.2

2020-08-16 @ 20:16:34 -- Running firmware released on: Dec 22 2019 02:19:57

2020-08-16 @ 20:16:34 -- Board: S411, version: 0

2020-08-16 @ 20:16:34 -- Unique device ID received - 0x24001f3337511138393639

2020-08-16 @ 20:17:53 -- couldn't retrieve version file due to internet availability

Suggest perhaps an "offline" mode for things like changing motor direction.   Other option is perhaps to run BLheli configurator, although not sure if that would pose problems with ESCs already flashed to JESC ?

[image]
[image]

lostcontrol, 2020-08-17:
I can confirm that. Wanted to flash some ESCs but my JESC on Linux is connecting but showing nothing (no read setup button, etc). First thought it was a Linux issue but couldn't access jflight.net from a browser and found that ticket.

Rogero-, 2020-08-17:
maybe he's doing some maintenance on the server, let's give it a day or two...

terrancefpv, 2020-08-17:
I can also confirm this issue.

@lostcontrol I was having that happen too but I realized I could get those buttons to show up if I just let it sit for a minute after connecting.

lostcontrol, 2020-08-17:
> @lostcontrol I was having that happen too but I realized I could get those buttons to show up if I just let it sit for a minute after connecting.

Thank you. I didn't have the patience to wait long enough I guess :wink: I want to convert a quad to 4.2 and RPM filtering so I need licenses anyway. Hope the service will be back online soon :crossed_fingers:

lostcontrol, 2020-08-17:
> maybe he's doing some maintenance on the server, let's give it a day or two...

Maintenance is usually something you announce so I guess something else is going on. I wrote in the JESC thread on rcgroups.com. I think it is @joelucid who is behind jflight.net but I'm not sure (I've been away from the quad stuff for some time).

chenrymotorsports, 2020-08-17:
I didnt see this thread when I joined this morning. I started a new thread immedeately after joining. Im experiencing the same issue. I upgraded a quad yesterday afternoon but went back yesterday evening to do another and the site couldnt be reached.

joelucid, 2020-08-18:
System is back up. There was an issue with automatic security updates which caused a hang. That's now also fixed.

closed 2020-08-18 by joelucid

jgauruder, 2020-08-18:
> System is back up. There was an issue with automatic security updates which caused a hang. That's now also fixed.

Is there value in decoupling the client side app from website reachability in some sort of "offline" mode?  Allow for using app to change motor directions or other parameter changes?