jflight-public/jesc-configurator #82: Hi, i cant enable bidirectional telemetry - Meteor65

David-VG, 2021-01-10:
I am trying to use jesc 48khz but I cant enable bidirectional telemetry:

[image]

[Log.txt](https://github.com/jflight-public/jesc-configurator/files/5793378/Log.txt)

[image]

I also noticed that gyro update frequency is readonly at 8khz

[image]

[image]

[BTFL_cli_Meteor65_20210110_231032.txt](https://github.com/jflight-public/jesc-configurator/files/5793379/BTFL_cli_Meteor65_20210110_231032.txt)

How can I get it working? All ESCs are licensed and uploaded both firmware + telemetry.

jake-walker, 2021-01-14:
I'm having the same issue today on a Meteor65, I found that it will still work fine without bidirectional DSHOT enabled. However, when it is enabled, I get 100% error on all the ESCs in Betaflight Configurator and all the sliders are stuck at the bottom, and the ESCs don't show up in JESC configurator unless I turn it off again.

After talking with some people about it, on the product page for the Meteor65 it says they ship with firmware `O_L_5_REV16_6.HEX` on the ESCs. I don't know whether there are different versions of the Meteor65 with different ESCs but it could be a possibility that we flashed O-H-5 firmware on to O-L-5 ESCs? I don't know what the differences are between L and H but I'm guessing it shouldn't be working at all for me (when I turn of bi-directional DSHOT) if that was the case.

David-VG, 2021-01-14:
> 
> 
> I'm having the same issue today on a Meteor65, I found that it will still work fine without bidirectional DSHOT enabled. However, when it is enabled, I get 100% error on all the ESCs in Betaflight Configurator, all the sliders are stuck at the bottom on the motors tab, I am missing the two or three notes at the end of the startup chime and the ESCs don't show up in JESC configurator unless I turn it off again and then it all goes back to normal.
> 
> After talking with some people about it, on the product page for the Meteor65 it says they ship with firmware `O_L_5_REV16_6.HEX` on the ESCs. Apparently later versions of the Meteor65 have H type but it could be a possibility that we flashed O-H-5 firmware on to O-L-5 ESCs? I don't know what the differences are between L and H but I'm guessing it shouldn't be working at all for me (when I turn of bi-directional DSHOT) if that was the case.

Mine finally worked. I've droped the original betaflight+config and started from scratch.

jake-walker, 2021-01-15:
Resetting fixed mine as well, it would be useful to know what might be causing the issues in the first place

odp-ilot, 2021-02-06:
Disable "SOFTSERIAL" in Betaflight-Configuration should work.