jflight-public/jesc-configurator #83: ESC #1 on crazybeeF4DX missing

CosmicChicken777, 2021-01-31:
I'm attached the log - tried flashing JESC and now ESC missing.  BL Heli won't see it either - followed instructions on JESC troubleshooting
[Log.txt](https://github.com/jflight-public/jesc-configurator/files/5899099/Log.txt)

CosmicChicken777, 2021-02-02:
@joelucid do you have any ideas?  Brand new board.  Trying to get JESC on it, I already paid the license fee, but then it lost the ESC #1 in the flashing process.

joelucid, 2021-02-03:
The log shows esc 1 missing even when you read the settings the first time. So it seems that it went missing during a flash attempt using blheli or one of its derivatives? JESC configurator has many flash stability related bug fixes and features which other blheli_s derivatives lack.

CosmicChicken777, 2021-02-03:
@joelucid I used a different computer for the original flash.  If I plugged it into the original computer and save the log, would that help you?

joelucid, 2021-02-03:
Yeah with the other log I can try to see what kind of failure occurred.

CosmicChicken777, 2021-02-07:
@joelucid here is the log from the other computer where the I first tried flashing with JESC.  Thanks!
[larva-jesc-Log3.txt](https://github.com/jflight-public/jesc-configurator/files/5939325/larva-jesc-Log3.txt)

CosmicChicken777, 2021-02-16:
> @joelucid here is the log from the other computer where the I first tried flashing with JESC. Thanks!
> [larva-jesc-Log3.txt](https://github.com/jflight-public/jesc-configurator/files/5939325/larva-jesc-Log3.txt)

@joelucid  Any luck looking at the log?  Is my Crazybeef4DX ruined?

CosmicChicken777, 2021-02-25:
@joelucid Hello?  You able to look at the previous log?